D3Engineering/d3-jetson-bsp #1: problems with .gitmodules

g-towboat, 2019-11-22:
Ive got some problems with the URLs in .gitmodules. After i change the URLs to:
` url = https://github.com/D3Engineering/ ` ....

it works for me

D3-growe, 2019-11-22:
Which branch are you working from? We did have problems with the submodule URLs but I believed that issue had been fixed.

I did a fresh clone and build of d3/2.0.0 and it was successful. Can you share your steps that failed?
```
git clone git@github.com:D3Engineering/d3-jetson-bsp.git test-clone
cd test-clone
git checkout d3/2.0.0
git submodule update --init
./bootstrap
./configure
make linux-defconfig
make -j8
```

g-towboat, 2019-11-25:
I work with the d3/2.0.0 branch.
After `git submodule update --init` iam getting a github access authorization error.

D3-growe, 2019-12-01:
Can you send me the exact error? I'm not able to reproduce the issue yet.

Thanks.

g-towboat, 2019-12-02:
`git clone git@github.com:D3Engineering/d3-jetson-bsp.git`
`Clone to 'd3-jetson-bsp' ...`
`ssh: connect to host github.com port 22: Connection timed out`
`fatal: Could not read from the remote repository.`

D3-growe, 2020-01-17:
I apologize for not responding sooner. We are looking to provide a better fix for this problem.