D3Engineering/d3-jetson-bsp #3: Sensor resolution /fps

g-towboat, 2020-01-17:
Why is a different resolution / FPS declared in the devicetree than in the data sheet on your website?
Are there any other device trees for the imx390?

https://github.com/D3Engineering/d3-jetson-bsp/blob/317c1de3e195baaae7cdaa66f4a4d1e3c5af2e1b/hardware/d3/templated-cameras/d3-cam-imx390-modes.dtsi#L25

D3-growe, 2020-01-17:
The HDR mode configuration we received from Sony does indeed operate at 60fps. Unfortunately we chose to disable the HDR mode for our 2.0.0 BSP release due to difficulties with HDR, particularly on Xavier. We are currently working with Nvidia and Sony to resolve the HDR issues.

You can re-enable the HDR mode by recompiling the kernel and dtb. HDR mode for the imx390 driver is a compile time option that can be selected using normal Linux kernel configuration.

g-towboat, 2020-01-22:
>  We are currently working with Nvidia and Sony to resolve the HDR issues.

When will the new version be released ?