D3Engineering/d3-jetson-bsp #7: nvgstcapture-1.0: fence issue

RomainNaour, 2020-11-03:
Hello,

I'm running nvgstcapture-1.0 just after booting the d3 rsp-tx2 board with a fpdlink camera connected to SIN2.
$ nvgstcapture-1.0

Everything seems ok and I get the video on the screen.
Then I stop the nvgstcapture-1.0 application.

Starting a second time fail and some message appear on dmesg

$ nvgstcapture-1.0

[11916.062618] host1x 13e10000.host1x: CaptureSchedule: syncpoint id 37 (15600000.isp_nvargus-daemon_4) stuck waiting 49422, timeout=-1

[11919.902610] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 33 (15600000.isp_nvargus-daemon_0) clientid 14134, HW thresh 14112, done 14114
[11919.915569] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 34 (15600000.isp_nvargus-daemon_1) clientid 14134, HW thresh 7056, done 7057
[11919.928349] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 35 (15600000.isp_nvargus-daemon_2) clientid 14134, HW thresh 7056, done 7057
[11919.941125] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 36 (15600000.isp_nvargus-daemon_3) clientid 14134, HW thresh 7056, done 7057
[11919.953898] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 38 (15600000.isp_nvargus-daemon_5) clientid 14134, HW thresh 7056, done 7057
[11919.966669] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 37 (15600000.isp_nvargus-daemon_4) clientid 14134, HW thresh 49419, done 49419

[11920.158597] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 47 (15700000.vi_2) clientid 14143, HW thresh 7056, done 7057
[11920.169988] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 46 (15700000.vi_1) clientid 14143, HW thresh 7056, done 7057
[11920.181376] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 45 (15700000.vi_0) clientid 14143, HW thresh 7139, done 7139

[11930.142614] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 33 (15600000.isp_nvargus-daemon_0) clientid 14134, HW thresh 14116, done 14118
[11930.155573] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 34 (15600000.isp_nvargus-daemon_1) clientid 14134, HW thresh 7058, done 7059
[11930.168351] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 35 (15600000.isp_nvargus-daemon_2) clientid 14134, HW thresh 7058, done 7059
[11930.181128] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 36 (15600000.isp_nvargus-daemon_3) clientid 14134, HW thresh 7058, done 7059
[11930.193902] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 38 (15600000.isp_nvargus-daemon_5) clientid 14134, HW thresh 7058, done 7059
[11930.206673] host1x 13e10000.host1x: cdma_handle_timeout: timeout: 37 (15600000.isp_nvargus-daemon_4) clientid 14134, HW thresh 49435, done 49435

It seems an issue with host1x (nvidia) driver that use sync_fence_wait() (from drivers/staging/android/sync.c !) related to CONFIG_TEGRA_GRHOST_SYNC kernel option and NVHOST32_IOCTL_CTRL_SYNC_FENCE_CREATE IOCTL.

I'm using the d3-jetson-3.0.1 SDK.
Can you reproduce this issue with an ar0820 fpdlink camera?

Best regards,
Romain

D3-growe, 2020-11-04:
Hi Romain,

What is your specific hardware configuration. You mentioned a Tx2 on a D3 FPDLink III RSP baseboard. Which ar0820 camera module are you using? Have you looked to see if this occurs with v4l2 in addition to Argus?

Thanks,
Greg

RomainNaour, 2020-11-04:
Hi Greg,

I'm using a ar0231 camera sensor from Leopard Imaging and I've ported a driver for this camera to d3-jetson 3.0.1 BSP.
This driver is based on the ar0820 available in d3-jetson BSP and the Leopard ar0231 driver for the Nvidia Tx2 development board.
There are some similarity between those two drivers and I tried to port the ar0231 on the d3 devicetree camera template.

The timeout occur after starting again the video streaming:

```
[11909.743742] ar0231_s_stream: start streaming.
[11911.390592] fence timeout on [ffffffc1e24a7c00] after 1500ms
[11911.390595] fence timeout on [ffffffc1b5b7b9c0] after 1500ms
[11911.390600] name=[nvhost_sync:33], current value=14112 waiting value=14113
[11911.390601] name=[nvhost_sync:36], current value=7056 waiting value=7057
[11911.390605] ---- mlocks ----
[11911.390605] ---- mlocks ----
[11911.390616] 8: locked by channel 7
[11911.390617] 8: locked by channel 7
[11911.390634] ---- syncpts ----
[11911.390634] ---- syncpts ----
[11911.390645] id 5 (disp_d) min 648 max 648 refs 1 (previous client : )
[11911.390649] id 6 (disp_e) min 7057 max 7057 refs 1 (previous client : )
[11911.390652] id 7 (disp_f) min 2 max 2 refs 1 (previous client : )
[11911.390656] id 8 (vblank1) min 713087 max -4 refs 1 (previous client : )
[11911.390673] id 22 (gp10b_507) min 13100 max 13100 refs 1 (previous client : )
[11911.390677] id 23 (gp10b_506) min 14 max 14 refs 1 (previous client : )
[11911.390680] id 24 (gp10b_505) min 3276 max 3276 refs 1 (previous client : gp10b_505)
[11911.390684] id 25 (15340000.vic_nvgstcapture-1._0) min 21153 max 21153 refs 1 (previous client : 15340000.vic_nvgstcapture-1._0)
[11911.390689] id 28 (15340000.vic_nvgstcapture-1._0) min 3 max 3 refs 1 (previous client : 15340000.vic_nvgstcapture-1._0)
[11911.390694] id 30 (15340000.vic_nvgstcapture-1._0) min 21150 max 21150 refs 1 (previous client : 15340000.vic_nvgstcapture-1._0)
[11911.390700] id 33 (15600000.isp_nvargus-daemon_0) min 14112 max 14116 refs 4 (previous client : 15600000.isp_nvargus-daemon_0)
[11911.390704] id 34 (15600000.isp_nvargus-daemon_1) min 7056 max 7058 refs 4 (previous client : 15600000.isp_nvargus-daemon_1)
[11911.390707] id 35 (15600000.isp_nvargus-daemon_2) min 7056 max 7058 refs 4 (previous client : 15600000.isp_nvargus-daemon_2)
[11911.390710] id 36 (15600000.isp_nvargus-daemon_3) min 7056 max 7058 refs 4 (previous client : 15600000.isp_nvargus-daemon_3)
[11911.390714] id 37 (15600000.isp_nvargus-daemon_4) min 49419 max 49430 refs 9 (previous client : 15600000.isp_nvargus-daemon_4)
[11911.390717] id 38 (15600000.isp_nvargus-daemon_5) min 7056 max 7058 refs 4 (previous client : 15600000.isp_nvargus-daemon_5)
...
```
I don't understand what all those log really means.

I also tried v4l2-compliance                                                                                                                                                                                                    
```
$ v4l2-compliance -d /dev/video0 
v4l2-compliance SHA   : not available

Driver Info:
        Driver name   : tegra-video
        Card type     : vi-output, ar0231 2-004c
        Bus info      : platform:15700000.vi:2
        Driver version: 4.9.140
        Capabilities  : 0x84200001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format

Compliance test for device /dev/video0 (not using libv4l2):
Required ioctls:
        test VIDIOC_QUERYCAP: OK

Allow for multiple opens:
        test second video open: OK
        test VIDIOC_QUERYCAP: OK
        test VIDIOC_G/S_PRIORITY: OK
        test for unlimited opens: OK

[... 46 lines cut ...]
                test VIDIOC_G_FMT: FAIL
                test VIDIOC_TRY_FMT: OK (Not Supported)
                test VIDIOC_S_FMT: OK (Not Supported)
                test VIDIOC_G_SLICED_VBI_CAP: OK (Not Supported)
                test Cropping: OK (Not Supported)
                test Composing: OK (Not Supported)
                test Scaling: OK (Not Supported)

        Codec ioctls:
                test VIDIOC_(TRY_)ENCODER_CMD: OK (Not Supported)
                test VIDIOC_G_ENC_INDEX: OK (Not Supported)
                test VIDIOC_(TRY_)DECODER_CMD: OK (Not Supported)

        Buffer ioctls:
                test VIDIOC_REQBUFS/CREATE_BUFS/QUERYBUF: OK
                fail: v4l2-test-buffers.cpp(574): VIDIOC_EXPBUF is supported, but the V4L2_MEMORY_MMAP support is missing, probably due to earlier failing format tests.
                test VIDIOC_EXPBUF: OK (Not Supported)

Test input 0:
Total: 43, Succeeded: 41, Failed: 2, Warnings: 2
```
Ok, there are some errors but I can't really test the d3 bsp using another camera to compare the result and check if the problem come from the ar0231 driver or from the Tegra stack.

Best regards,
Romain

D3-growe, 2020-11-04:
To stream using the v4l2 path use `v4l2-ctl --stream-mmap`. This path is usually easier to bring up than the Argus path. 

Also try enabling the trace logs (see here: https://elinux.org/Jetson_TX2_Camera_BringUp). Check to see that your camera really is supplying CSI frames on your second call to s_stream().

Regards,
Greg

RomainNaour, 2020-11-04:
On the first run v4l2-ctl --stream-mmap is ok

```
$ v4l2-ctl --stream-mmap
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 34.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 33.83 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 33.77 fps

```
But not on the second run:

```
[  148.775122] ar0231_s_stream: start streaming.
[  149.104182] tegra-vi4 15700000.vi: PXL_SOF syncpt timeout! err = -11
[  149.110794] tegra-vi4 15700000.vi: tegra_channel_error_recovery: attempting to reset the capture channel
....
[  163.641797] ar0231_s_stream: stop streaming.
[  163.745780] ar0231_power_off: power off.
```

Following the elinux wiki:

     kworker/0:3-3392  [000] ....  1556.842812: rtos_queue_peek_from_isr_failed: tstamp:52622718146 queue:0x0b4b4500
     kworker/0:3-3392  [000] ....  1557.010833: rtos_queue_peek_from_isr_failed: tstamp:52627718148 queue:0x0b4b4500
     kworker/0:3-3392  [000] ....  1557.122813: rtos_queue_peek_from_isr_failed: tstamp:52632718171 queue:0x0b4b4500

=> "Didn't receive any package from the MIPI bus. Make sure the sensor have output data to the MIPI bus"

What do you think ?
Writing 0x001C to 0x301A register is not enough to resume the video streaming ? We need to do something else ?

Best regards,
Romain

D3-growe, 2020-11-05:
Based on this evidence it strongly suggests to me that something isn't happening properly with your camera module to get it streaming again. The lack of logging in the trace log suggests that no data is being received on the Jetson.

RomainNaour, 2020-11-05:
Hello,

Thanks for the feedback.

I'm looking at the ar0231 driver provided by Leopard and the ar0820 driver by d3, why the "struct tegracam_device" is not used in d3 camera video drivers ? It seems all other video camera driver are using it.

On the ar0231 driver from Leopard, there are some undocumented gpios (pwdn_gpio, af_gpio, reset_gpio) used by the driver for power_on and power_off functions. I don't have the schematic of the Leopard camera...
It seems I need to control gpios available on ar0231 camera video sensor to resume the video streaming properly.
Why there are not handled by the d3 ar0820 driver ?

Best regards,
Romain

D3-growe, 2020-11-05:
Camera drivers are specific the camera modules which usually include a pairing of a serializer and image sensor. Every camera driver is unique, though they share many similar features.

The ar0820 driver in 3.0.1 is experimental. It is an early draft.