D3Engineering/d3-jetson-bsp #8: Rebuilding Device Tree with Kernel

MaximumProgrammer, 2020-11-04:
If the the device tree is changed in hardware/d3/* how can these files (templates) for camera can be used if also the the debian package is rebuild ?

Kind regards

cxw42, 2020-11-04:
On the target:
- Install the `d3-jetson-util` deb (that deb is made along with the linux-image deb)
- Run `d3-install-dtb`, and it will prompt you for which kernel, and which dtb.  It will then flash that dtb.

See doc/D3-Jetson-BSP-Release-Notes.md, section "DTB File Selection".

Hope this helps!

MaximumProgrammer, 2020-11-05:
Hello and thx,

there is the directory     ./hardware/d3/
                                     ./hardware/d3/templated-cameres

and so on, i mainly want to add aditional drivers for our camera with using the templates from d3, how is going on in this case the build and deploy proces ?

Kind regards.

cxw42, 2020-11-05:
cd to `hardware/d3`, then:
1. copy the top-level dts into a new file.  E.g., `cp d3-xavier-2xocto.dts FOO.dts`
1. add the corresponding dtb to the build: in `Makefile`, add `d3-dtb-y += FOO.dtb`
1. build, and make sure the .dtb file was generated into `build/kernel/arch/arm64/boot/dts/FOO.dtb`.
1. Back in `FOO.dts`, follow the `#include`-file chain down to `templated-cameras`.  E.g., for `d3-xavier-2xocto.dts`, `hardware/d3/d3-2xocto-cameras.dtsi` is what actually `#include`s the `templated-cameras/d3-cam-....dtsi` files.
1. Add another `#include` there for your new `templated-cameras/BAR.dtsi` file.

MaximumProgrammer, 2020-11-05:
Ok thx

MaximumProgrammer, 2020-11-06:
> cd to `hardware/d3`, then:
> 
>     1. copy the top-level dts into a new file.  E.g., `cp d3-xavier-2xocto.dts FOO.dts`
> 
>     2. add the corresponding dtb to the build: in `Makefile`, add `d3-dtb-y += FOO.dtb`
> 
>     3. build, and make sure the .dtb file was generated into `build/kernel/arch/arm64/boot/dts/FOO.dtb`.
> 
>     4. Back in `FOO.dts`, follow the `#include`-file chain down to `templated-cameras`.  E.g., for `d3-xavier-2xocto.dts`, `hardware/d3/d3-2xocto-cameras.dtsi` is what actually `#include`s the `templated-cameras/d3-cam-....dtsi` files.
> 
>     5. Add another `#include` there for your new `templated-cameras/BAR.dtsi` file.

Ok really thank you, i think this will help me.

Kind regards.

cxw42, 2020-11-06:
Glad to be able to help!

MaximumProgrammer, 2020-11-10:
> Glad to be able to help!

Ok but there is still one question how can i force to rebuild all dtb files which are in the directory ./hardware/d3/ because i am sure that the build process is not regenerationg all dtb files even when i am changing or adding some files there is still one old build id or build time for the device tree, on the opposite for the kernel i am gettting different build ids, i just wanted to build the a debian package with make bin-kernel but on the jetson i am getting different build ids.

for example 

cat /proc/device-tree/d3,jetson-bsp-build-id

3fbaf032-0dbf-46cc-9277-f3151b45dc1e

and

cat /sys/module/build_id/version 

44a62126-d1ce-4424-bd8b-b45e0455b816

Kind regards

cxw42, 2020-11-12:
We've had problems with mismatches before, so that might be a known issue --- I'm not sure off-hand.  Check README.md and doc/Release-Notes.md (generated).  
- If you are using `make sync`, the kernel and dtb build-ids should match.  
- If you are using debs, they will not match until you run `d3-install-dtb`.  
- Finally, if /boot/extlinux/extlinux.conf lists a kernel other than the one you just installed, they won't match.  E.g., if you install from debs and then switch to `make sync`, you will need to edit extlinux.conf to change the image filename from `/boot/vmlin...` to `/boot/Image`.

MaximumProgrammer, 2020-11-12:
> We've had problems with mismatches before, so that might be a known issue --- I'm not sure off-hand. Check README.md and doc/Release-Notes.md (generated).
>
>     * If you are using `make sync`, the [...]

I think the main problem was just that if the device tree compiler fails then there is not exception during the building process, the process still proceeds even the device tree was not able to rebuild, so the script should stop at this point. I guess this mainly because there is no exception handling afterwards all files (recompiled files) are moved into some directories for the debian package.

cxw42, 2020-11-13:
That is strange.  If you can reproduce dts failures not halting the build, could you please open a new bug with the full log of a run that shows the problem?  I have never encountered that issue.

I have seen other files continue to build because of `-j` even after dtc has failed, but the build always eventually terminates with an error in that case, in my experience.

Remember, use either debs or `make sync`, but don't switch between them unless you very carefully check extlinux.conf.

MaximumProgrammer, 2020-11-13:
I was able to reproduce the failure because i changed some files in the templates of the device tree, and i saw the device tree build fails, after using the original files from the device tree, the script or build process proceeds with also changing and updating the device tree. Maybe this because building a debian package only copies some files. I also an error that he was not able to build the tree but the script still proceeds.

I used make bin-deb for building a whole package.

cxw42, 2020-11-13:
Definitely odd.

For debs, we use `make release-zip` or `bin/jenkins-build`.  For just the kernel deb, try `make bin-kernel`.

MaximumProgrammer, 2020-11-13:
Thx but i am already using make bin-kernel, but i think he is skipping the device tree compiling even when it fails.

cxw42, 2020-11-13:
Sorry you're having trouble!  As I say, please open a new bug with full instructions to reproduce, and the log of your run, showing that the build succeeds even when dtc(1) has failed in the middle.  I'll take a look.  Thanks!

MaximumProgrammer, 2020-11-14:
Ok i will do, i don't consider this really as bug.

closed 2020-11-16 by D3-growe