D3Engineering/d3-jetson-bsp #10: Probe function not called

MaximumProgrammer, 2020-11-13:
Hello i want to write my own camera driver, i changed the device tree and also some parts of the driver. Nevermind i also added some debug functions to the original driver code of the camera ov10640 just for example the lines :

pr_info("ov10640 probe");
printk(KERN_ALERT "DEBUG: Passed %s %d \n",__FUNCTION__,__LINE__);
dev_dbg(&client->dev, "ov10640 probe enter");

because i wan to to see if the overlay loader respective the kernel of the jetson xavier is executing the function, i did't not see any message but i also execute the line 

echo "7" > /proc/sys/kernel/printk

so for example i must see the line 

pr_info("ov10640 probe");

in the kernel messages (dmesg) because i also see some informations from the overlay-loader.

The probe function still has the body:

static int ov10640_probe(struct i2c_client *client,
		const struct i2c_device_id *id)
{
	int err = 0;
	struct camera_common_data *common_data;
	struct ov10640 *self = NULL;
	int tries;

	pr_info("ov10640 probe");
	printk(KERN_ALERT "DEBUG: Passed %s %d \n",__FUNCTION__,__LINE__);
	dev_dbg(&client->dev, "ov10640 probe enter");

	TRY_MEM(common_data, devm_kzalloc(&client->dev, sizeof(*common_data),
				GFP_KERNEL));
	TRY_MEM(self, devm_kzalloc(&client->dev, sizeof(*self), GFP_KERNEL));
	self->client = client;
	self->dev = &client->dev;

	common_data->priv = self;
	common_data->ops = &ov10640_common_ops;
	common_data->ctrl_handler = &self->ctrl_handler;
	common_data->dev = &client->dev;
	common_data->frmfmt = &ov10640_formats[0];
	common_data->numfmts = ov10640_formats_len;
	common_data->colorfmt = camera_common_find_datafmt(MEDIA_BUS_FMT_SBGGR12_1X12);

	common_data->power = &self->power;
	common_data->ctrls = self->ctrls;

	common_data->priv = self;
	common_data->numctrls = ov10640_ctrls_count();

	common_data->def_mode = common_data->frmfmt[OV10640_MODE_DEFAULT].mode;
	common_data->def_width = common_data->frmfmt[OV10640_MODE_DEFAULT].size.width;
	common_data->def_height =
		common_data->frmfmt[OV10640_MODE_DEFAULT].size.height;
	/* common_data->def_clk_freq = 48000000; */
	common_data->fmt_width = common_data->def_width;
	common_data->fmt_height = common_data->def_height;

	self->s_data = common_data;
	self->s_data->sensor_mode_id = OV10640_MODE_DEFAULT;
	self->s_data->mode = OV10640_MODE_DEFAULT;
	self->subdev = &common_data->subdev;
	self->subdev->dev = self->dev;
	self->s_data->dev = self->dev;

	self->pdata = ov10640_parse_dt(self, common_data);
	if (!self->pdata)
		return -EFAULT;

	TRY_MEM(self->map, devm_regmap_init_i2c(self->client, &ov10640_regmap_cfg));

	for (tries = 50; --tries >= 0; ) {
		/* regmap_write(self->map, OV10640_REG_SOFTWARE_CTRL2, */
		/*	     OV10640_REG_SOFTWARE_CTRL2_RESET); */

		err = regmap_multi_reg_write(self->map, mode_table[OV10640_MODE_DEFAULT].reg_sequence, mode_table[OV10640_MODE_DEFAULT].size);

		if (err >= 0)
			err = regmap_write(self->map, OV10640_REG_SOFTWARE_CTRL1, OV10640_REG_SOFTWARE_CTRL1_SW_STBY);

		if (err < 0) {
			dev_dbg(self->dev, "Giving device more time to settle\n");
			usleep_range(50 * 1000, 50 * 1000);
		} else
			break;
	}
	if (err < 0) {
		dev_err(self->dev, "Failed to find device\n");
		return err;
	}

	TRY(err, camera_common_initialize(common_data, "ov10640"));

	v4l2_i2c_subdev_init(self->subdev, client, &ov10640_subdev_ops);
	TRY(err, ov10640_ctrls_init(self));

	self->subdev->internal_ops = &ov10640_subdev_internal_ops;
	self->subdev->flags |= V4L2_SUBDEV_FL_HAS_DEVNODE | V4L2_SUBDEV_FL_HAS_EVENTS;

	TRY(err, ov10640_media_init(self));
	TRY(err, v4l2_async_register_subdev(self->subdev));

	dev_dbg(self->dev, "probe success");
	return 0;
}

....

wheras the compatible is just:

static struct of_device_id ov10640_of_match[] = {
	{ .compatible = "d3,ov10640"},
	{ },
};

.....

In the extlinux.conf i also added the lines

active_overlays=ov10640_0,ov10640_1,ov10640_2,ov10640_3,ov10640_4,ov10640_5

for activating the drive, the driver tree now looks like for example:

		mux@70 {
			status = "okay";
			compatible = "nxp,pca9546";
			reg = <0x70>;
			#address-cells = <0x1>;
			#size-cells = <0x0>;
			vcc-supply = <0x2d>;
			vcc-pullup-supply = <0x2d>;

			i2c@0 {
				status = "okay";
				reg = <0x0>;
				#address-cells = <0x1>;
				#size-cells = <0x0>;
				i2c-mux,deselect-on-exit;
				linux,phandle = <0x1a5>;
				phandle = <0x1a5>;

				ub960@32 {
					status = "okay";
					compatible = "d3,ub960";
					reg = <0x32>;
					#address-cells = <0x1>;
					#size-cells = <0x0>;
					iovdd-supply = <0x2d>;
					avdd-supply = <0x39>;
					pdb-gpios = <0x13 0x9e 0x1>;
					csi-tx-speed-mbps = <0x640>;
					csi-lane-count = <0x4>;
					csi-continuous-clock = <0x1>;
					linux,phandle = <0x3a>;
					phandle = <0x3a>;

					link@0 {
						status = "okay";
						reg = <0x0>;
						#address-cells = <0x1>;
						#size-cells = <0x0>;
						linux,phandle = <0x1a6>;
						phandle = <0x1a6>;

						ub953@30 {
							compatible = "d3,ub953";
							status = "disabled";
							physical-addr = <0x30>;
							#address-cells = <0x1>;
							#size-cells = <0x0>;
							reg = <0x30>;
							csi-lane-count = <0x4>;
							csi-continuous-clock = <0x1>;
							i2c-voltage-sel = <0x0>;
							linux,phandle = <0x14b>;
							phandle = <0x14b>;

							ov10640@20 {
								status = "disabled";
								compatible = "d3,ov10640";
								#address-cells = <0x1>;
								#size-cells = <0x0>;
								reg = <0x20>;
								physical-addr = <0x32>;
								devnode = "video0";
								mclk = "extperiph1";
								physical_w = "5.41";
								physical_h = "4.57";
								sensor_model = "ov10640";
								use_decibel_gain = "false";
								use_sensor_mode_id = "true";
								deserializer = <0x3a>;
								linux,phandle = <0x149>;
								phandle = <0x149>;

								mode0 {
									mclk_khz = "25000";
									mclk_multiplier = "22.0";
									num_lanes = [34 00];
									tegra_sinterface = "serial_a";
									discontinuous_clk = "no";
									dpcm_enable = "false";
									cil_settletime = [30 00];
									csi_pixel_bit_depth = "12";
									pixel_phase = "bggr";
									active_w = "1280";
									active_h = "1080";
									readout_orientation = [30 00];
									line_length = "1450";
									inherent_gain = [31 00];
									serdes_pix_clk_hz = "535000000";
									pix_clk_hz = "71428571";
									min_framerate = "30";
									max_framerate = "30";
									embedded_metadata_height = [30 00];
									vc_id = [30 00];
									mode_type = "bayer";
									min_gain_val = "1.0";
									max_gain_val = "8.0";
									min_exp_time = "20";
									max_exp_time = "33698";
									min_hdr_ratio = [31 00];
									max_hdr_ratio = [31 00];
								};

								ports {
									#address-cells = <0x1>;
									#size-cells = <0x0>;
									status = "disabled";

									port@0 {
										reg = <0x0>;

										endpoint {
											vc-id = <0x0>;
											status = "disabled";
											port-index = <0x0>;
											bus-width = <0x4>;
											remote-endpoint = <0x2e>;
											linux,phandle = <0x14a>;
											phandle = <0x14a>;
										};
									};
								};
							};

In the device tree there is still this node for ov10640 for activating the corresponding driver should it be renamed to d3_ov10640_0 in the boot/extlinux/extlinux.conf ? to see the probe function is called ? and how can i check if the the overlay-loader realy tries to start the driver from getting some statuse ? is there any was to get some feedback like with checking the status and so on. Is there a problem with the overlay loader ? The probe function should be always called and it must be called to start streaming of the camera, i didn't changed any part of the original code, i only added some debug informations in the probe function.

Kind regards

cxw42, 2020-11-13:
- Run `make linux-defconfig` and then `make linux-menuconfig` --- make sure the overlay-loader driver is enabled
- In the kernel command line, enable overlay_loader dyndbg, and it will log what it is trying.
- I think your `active_overlays` is correct.

MaximumProgrammer, 2020-11-13:
Ok i am seeing that the overlay loader is matching cameras but afterwards i am getting no debug infos from my probe function, does it mean anything is wrong with my device tree ? just what can be the reason not calling any probe function ?

D3-growe, 2020-11-13:
Markus,

It may help you to enable dyanmic debug for the driver. Pass ov10640.dyndbug="+plmf" as an argument to the kernel via extlinux.conf. You will then see all of the `dev_dbg()` (et. al.) debug output.

Regards,
Greg

Please be aware that this email includes email addresses outside of the organization.

MaximumProgrammer, 2020-11-14:
Ok thx, this will help a lot, so it means the overlay loader and the camera have different debug modes, and the overlay loader ist perhaps hiding some debug informations.

D3-growe, 2020-11-15:
No, not exactly. Overlay loader and the camera drivers are different modules. As different modules you can set different dynamic debug settings for each.

MaximumProgrammer, 2020-11-26:
I think this thread can be closed. 

Kind regards.

closed 2020-11-30 by D3-growe