D3Engineering/d3-jetson-bsp #12: Support for FPD Link iii.

AjithKumaresh4, 2021-02-18:
Hi Team,

I am trying to bring up the D3RCM-OV10640-953 Camera Module in the Jetson Xavier NX board with DS90UB954 deserializer. I am using the following drivers  (https://github.com/D3Engineering/d3-jetson-bsp/tree/d3/3.0.1/kernel/d3/drivers/d3). The Ser/Des connection is up and the sensor registered the /dev/video0 node. But when I issued the camera preview command I got the following error. 
**[ +2.479975] tegra194-vi5 15c10000.vi: no reply from camera processor
[ +0.000151] tegra194-vi5 15c10000.vi: uncorr_err: request timed out after 2500 ms**

Could you please give some pointers to solve this issue, Does the following driver is enough to add support for the D3RCM-OV10640-953 Camera Module (https://github.com/D3Engineering/d3-jetson-bsp/tree/d3/3.0.1/kernel/d3/drivers/d3)?

Thanks,
Ajith.

D3-growe, 2021-02-18:
Hi Ajith,

We don't support the our NX carrier board with the 3.0.1 release. Can you please update to the 4.0.0 BSP? Your configuration should be fully supported with that release. 

Thanks,
Greg

AjithKumaresh4, 2021-02-19:
Hi Team,
I have moved to the 4.0.0 BSP release And I tested the FPD Link iii. Still, I am facing the same error.

I have the following doubts,
As per this document, GPIO_0 and GPIO_1 are connected to the Reset and Powerdown pins of the sensors. - [().](https://d3engineering.com/assets/uploads/data-sheets/camera-modules/D3Eng-DesignCore-D3RCM-OV10640-953-DataSheet.pdf) 

1) Should I use Serializer GPIO to reset the sensor or doing a digital reset on the sensor is enough(writing RESET register from sensor driver)?

I used the Ser/Des and sensor templates to create our own device tree file for the FPD module. [()](https://github.com/D3Engineering/d3-jetson-bsp/tree/d3/4.0.0/hardware/d3)

2) If I use the following serializer dts configurations as per the serializer template
gpio-rmten = <0x0>; 
gpio-out-src = <0x6>;
gpio-out-en = <0x3>; 
gpio-in-en = <0x4>;
It sets the Serializer GPIOs as **GPIO_0-OUTPUT-HIGH** and **GPIO_1-INPUT-HIGH**. Both should be in **OUTPUT-HIGHT** state right?. 
Please correct me If am wrong.

3) I want to use a single FPD camera without external fsync support then Can I disable the fsync support or should I use the internal fsync support?

My device tree configurations 
ds90ub954: ub954@3d {
        status = "okay";
        compatible = "d3,ub960";
        reg = <0x3d>;
        #address-cells = <1>;
        #size-cells = <0>;
        iovdd-supply = <&p3668_vdd_1v8_sd>;
        pdb-gpios = <&tegra_main_gpio TEGRA194_MAIN_GPIO(Z, 3) GPIO_ACTIVE_HIGH>;
        csi-tx-speed-mbps = <1600>;
        csi-lane-count = <2>;
        csi-continuous-clock = <1>;

        ds90ub954_link0: link@0 {
                status = "okay";
                reg = <0>;
                #address-cells = <1>;
                #size-cells = <0>;

                ub953_0: ub953@3c {
                        compatible = "d3,ub953";
                        status = "okay";
                        physical-addr = <0x18>;
                        #address-cells = <1>;
                        #size-cells = <0>;
                        reg = <0x3c>;
                        csi-lane-count = <2>;
                        csi-continuous-clock = <1>;
                        i2c-voltage-sel = <0x0>;

                        gpio-rmten = <0x0>;
                        gpio-out-src = <0x6>;
                        gpio-out-en = <0x6>;
                        gpio-in-en = <0x0>;
                        hs-clk-div = <2>;
                        div-m-val = <1>;
                        div-n-val = <40>;
                        wait-for-self-configure;

                        ov10640_0: ov10640@40 {
                                status = "okay";
                                compatible = "d3,ov10640";
                                #address-cells = <1>;
                                #size-cells = <0>;
                                reg = <0x40>;
                                physical-addr = <0x32>;
                                devnode = "video2";
                                mclk = "extperiph1";
                                physical_w = "5.41";
                                physical_h = "4.57";
[... 37 lines cut ...]
                                };

                                ports {
                                        #address-cells = <1>;
                                        #size-cells = <0>;
                                        status = "okay";

                                        port@0 {
                                                reg = <0>;
                                                sen_out_ov10640_0: endpoint {
                                                        vc-id = <0>;
                                                        status = "okay";
                                                        port-index = <1>;
                                                        bus-width = <2>;
                                                        remote-endpoint = <&ov10640_csi_in2>;
                                                };
                                        };
                                };
                        };
                };
         };
};

Thanks,
Ajith.

AjithKumaresh4, 2021-02-19:
> Hi Team,
> I have moved to the 4.0.0 BSP release And I tested the FPD Link iii. Still, I am facing the same error.
> 
> I have the following doubts,
> As per this document, GPIO_0 and GPIO_1 are connected to the Reset and Powerdown pins of the sensors. - [().](https://d3engineering.com/assets/uploads/data-sheets/camera-modules/D3Eng-DesignCore-D3RCM-OV10640-953-DataSheet.pdf) 
> 
> 1. Should I use Serializer GPIO to reset the sensor or doing a digital reset on the sensor is enough(writing RESET register from sensor driver)?
> 
> I used the Ser/Des and sensor templates to create our own device tree file for the FPD module. [()](https://github.com/D3Engineering/d3-jetson-bsp/tree/d3/4.0.0/hardware/d3)
> 
> 1. If I use the following serializer dts configurations as per the serializer template
>    gpio-rmten = <0x0>;
>    gpio-out-src = <0x6>;
>    gpio-out-en = <0x3>;
>    gpio-in-en = <0x4>;
>    It sets the Serializer GPIOs as **GPIO_0-OUTPUT-HIGH** and **GPIO_1-INPUT-HIGH**. Both should be in **OUTPUT-HIGHT** state right?.
>    Please correct me If am wrong.
> 2. I want to use a single FPD camera without external fsync support then Can I disable the fsync support or should I use the internal fsync support?
> 
> My device tree configurations
> ds90ub954: ub954@3d {
> status = "okay";
> compatible = "d3,ub960";
> reg = <0x3d>;
> #address-cells = <1>;
> #size-cells = <0>;
> iovdd-supply = <&p3668_vdd_1v8_sd>;
> pdb-gpios = <&tegra_main_gpio TEGRA194_MAIN_GPIO(Z, 3) GPIO_ACTIVE_HIGH>;
> csi-tx-speed-mbps = <1600>;
> csi-lane-count = <2>;
> csi-continuous-clock = <1>;
> 
> ```
>     ds90ub954_link0: link@0 {
>             status = "okay";
>             reg = <0>;
>             #address-cells = <1>;
>             #size-cells = <0>;
> 
>             ub953_0: ub953@3c {
>                     compatible = "d3,ub953";
>                     status = "okay";
>                     physical-addr = <0x18>;
>                     #address-cells = <1>;
>                     #size-cells = <0>;
>                     reg = <0x3c>;
>                     csi-lane-count = <2>;
>                     csi-continuous-clock = <1>;
>                     i2c-voltage-sel = <0x0>;
> 
>                     gpio-rmten = <0x0>;
>                     gpio-out-src = <0x6>;
>                     gpio-out-en = <0x6>;
>                     gpio-in-en = <0x0>;
>                     hs-clk-div = <2>;
>                     div-m-val = <1>;
>                     div-n-val = <40>;
>                     wait-for-self-configure;
> 
>                     ov10640_0: ov10640@40 {
>                             status = "okay";
>                             compatible = "d3,ov10640";
>                             #address-cells = <1>;
>                             #size-cells = <0>;
>                             reg = <0x40>;
>                             physical-addr = <0x32>;
>                             devnode = "video2";
>                             mclk = "extperiph1";
>                             physical_w = "5.41";
>                             physical_h = "4.57";
[... 39 lines cut ...]
>                             ports {
>                                     #address-cells = <1>;
>                                     #size-cells = <0>;
>                                     status = "okay";
> 
>                                     port@0 {
>                                             reg = <0>;
>                                             sen_out_ov10640_0: endpoint {
>                                                     vc-id = <0>;
>                                                     status = "okay";
>                                                     port-index = <1>;
>                                                     bus-width = <2>;
>                                                     remote-endpoint = <&ov10640_csi_in2>;
>                                             };
>                                     };
>                             };
>                     };
>             };
>      };
> ```
> 
> };
> 
> Thanks,
> Ajith.

D3-growe, 2021-02-19:
Ajith,

You shouldn't need to modify any code. We have tested the binary build in the configuration that you are working with. The basic installation procedure is:

1. Follow the directions in the NX bring-up document in the doc/D3-Jetson-NX-Carrier-Bringup.md
2. Install Jetpack 4.4 (please use this exact JetPack version)
3. Install the D3 .deb packages for the kernel and utilities
4. Install the D3 DTB
5. Select your camera model and position (d3-select-cameras-boot)

Please consult the NX Carrier Bringup document for the exact installation details. If you follow these directions you should not need to modify and code.

Thanks,
Greg

AjithKumaresh4, 2021-02-22:
Hi @D3-growe 

Thanks for your response.

I am trying to bring up the following camera module (https://d3engineering.com/assets/uploads/data-sheets/camera-modules/D3Eng-DesignCore-D3RCM-OV10640-953-DataSheet.pdf) in our custom board with UB954 deserializer. Our design uses UB953/UB954 Ser/Des connection for FPD Link iii.

I took the UB960, UB953, and OV10640 drivers as it is from the following sources.(https://github.com/D3Engineering/d3-jetson-bsp/tree/d3/4.0.0/kernel/d3/drivers/d3) And I modified the dts based on our design. The drivers are up and /dev/video0 node got created. But the preview doesn't work.

I have the following doubts could you please clarify these.
     1) Should I use Serializer GPIO to reset the sensor or doing a digital reset on the sensor is enough(writing RESET register from sensor driver)?
     2) If I use the following serializer dts configurations as per the serializer template
gpio-rmten = <0x0>;
gpio-out-src = <0x6>;
gpio-out-en = <0x3>;
gpio-in-en = <0x4>;
It sets the Serializer GPIOs as GPIO_0-OUTPUT-HIGH and GPIO_1-INPUT-HIGH. Both should be in OUTPUT-HIGHT state right?.
Please correct me If am wrong.
     3) I want to use a single FPD camera without external fsync support then Should I disable the fsync support or use the internal fsync support?

Thanks,
Ajith.

D3-growe, 2021-02-22:
Ajith,

I see. If I understand correctly you are asking advice about custom hardware, not about D3 hardware.

1. Yes, if your imager has a reset pin should should operate that using the serializer. The best approach would be to expose the serializer as a GPIO controller in Linux. If you make those changes we'd love to get a pull request!
2. Without looking at the docs for the ub954 I can't be certain. Those attribute, basically, directly map into what you'll see documented in the datasheets for the TI ub954.
3. If you don't care about frame sync then you should disable it.

I'd like to resolve this issue. I think that you'll get better feedback if you post on NVIDIA's devtalk forum with these questions -- you'll have a much larger audience.

Thanks,
Greg

closed 2021-02-22 by D3-growe