D3Engineering/d3-jetson-bsp #13: Set up os support on target

ganindu7, 2021-05-06:
Hi I'm not familiar with automake and I'm trying to understand [this](https://github.com/ganindu7/d3-jetson-bsp/tree/d3/4.0.0#set-up-os-support-on-the-target)

```

#!/bin/bash
# @configure_input@

TARGET="@TARGET_HOST@"

ssh-copy-id @USERNAME_AT@"$TARGET"
ssh @USERNAME_SSH@ -t "$TARGET" 'sudo apt-get -y update && sudo apt-get install -y rsync'
make sync-debug
ssh @USERNAME_SSH@ -t "$TARGET" 'cd debug; sudo ./prep-new-debug-system'
ssh @USERNAME_SSH@ -t "$TARGET" 'debug/disable-screen-lock'
ssh @USERNAME_SSH@ -t "$TARGET" 'cd debug; ./build-tm-api'
```

To run this file I think I need to fill the @...@'s and I'm just wondering what's the best way to do this. 

Do I need to do something before I run `bin/prep-new-system-from-host`

Cheers.

D3-growe, 2021-05-06:
Hi,

This script prepares the Jetson for D3's development tools. If you are just wishing to deploy the D3 BSP you do not need to do any of this. These steps are only recommended if you are developing kernel code. These are the steps and tools that we use internally at D3.

It looks like our README.md is a bit out of order. I think this is because the "prep script" didn't used to be an autotools template.

What you should do is run the configure script which is documented in the next section of the README.md.

The tl;dr; steps for setting up a development environment are the following:
```
./bootstrap
./configure
bin/prep-new-system-from-host
make -j8
make sync && make reboot
```

regards,
Greg

ganindu7, 2021-05-06:
Hi, 
Thanks a lot @D3-growe I'm trying to do a source build, I'm not familiar with kernel development but I like to understand what's going on haha. 
Cheers, 
Ganindu.

closed 2021-05-06 by ganindu7