D3Engineering/d3-jetson-bsp #14: .dts file missing from submodules? (when making the kernel)

ganindu7, 2021-05-26:
Hi, I'm trying to make the kernel but it seems a file is missing from the submodules? 
However I'm not sure if I've understood the error fully. I've linked the [make output](https://github.com/D3Engineering/d3-jetson-bsp/files/6547166/make_output.txt) file for clues. @D3-growe (not sure if you're the right person)

Cheers,
Ganindu.

D3-growe, 2021-05-26:
Hi Ganindu,

Which branch are you building?

There's a "chicken/egg" problem when building the D3 BSP. We have some logic that expects the .dtb file to exist when you run configure but it can't exist before it's built. It's a known issue. The good news is that you can work around it pretty easily.

I recommend running configure with no arguments first. Then build. Then run configure with the arguments that you really want (--with-dtb, --with-system-type, etc). 

If you let me know which branch you are on I'm sure I can help you. The error listed in your make output can happen when moving from a newer branch of code to an older branch. What happens is the jakku submodule files linger and cause a problem with building due to the use of wildcards. If that's actually the case we can very easily fix it.

Thanks,
Greg

ganindu7, 2021-05-26:
Hi Greg, 
I'm on  d3/4.0.0 I think the main cause is the submodules not being pulled in (`git submodule update --recursive --init` command not working) 

basically my `d3-jetson-bsp/hardware/nvidia/platform/t19x/jakku`  is empty.

D3-growe, 2021-05-26:
What isn't working for the submodule update? Do we (D3) have the wrong submodule URL for jakku perhaps?

ganindu7, 2021-05-26:
I'm not really sure. That may be the case because when I initially ran it, it did download some stuff but not the stuff in the `jaku` directory. but I can follow the link and end up in a non empty folder when using github. (I'm not also a got expert I just followed the instructions.

ganindu7, 2021-05-27:
@D3-growe Apologies! I was slightly confused because This could be a combination of IT issues as well. If it helps I've added some logs in case they are helpful. 
[submodule_update_out.txt](https://github.com/D3Engineering/d3-jetson-bsp/files/6553983/submodule_update_out.txt)
[submodule_update_out_full.txt](https://github.com/D3Engineering/d3-jetson-bsp/files/6553984/submodule_update_out_full.txt)

D3-growe, 2021-05-27:
No apologies are needed! I've got an engineer looking at this right now. So far there are two problems:

1. The README in d3/4.0.0 is out of order. You can't run bin/prep-new-debug-system-from-host until `configure` is run which is done in the steps that follow.
2. Something isn't right with the `jakku` submodule as you've noticed. I think this can be fixed by editing `l4t-submodules/.gitmodules` but I'm waiting for verification.

Thanks,
Greg

D3-growe, 2021-05-27:
Ganindu,

I believe I found the source of the issue. The jakku submodule was using a fully qualified URL that used ssh. Therefore, if you cloned the main repo using http and your ssh key wasn't associated with any github account, then the jakku submodule clone would fail. I've changed the submodule to be a relative URL, like the others, in `d3/4.0.0`. With a relative URL the same authentication/authorization mechanism will be used for submodule as is used by the parent repository.

I also updated the README to reorder sections as it was confusing and, if followed exactly, would not work.

Please update to the d3/4.0.0 branch (do not use the D3/RELEASE/4.0.0 tag). I believe this branch will solve these issues.

Thanks,
Greg

ganindu7, 2021-05-29:
Hi @D3-growe  Thanks a lot!!! (I haven't tried it just yet and will update you ASAP)

I got one off topic question

 Just wondering if this can work with jetpack 4.5.1 or (the latest at any time)  or should I strictly be using 4.4 ? 

Cheers,
Ganindu.

D3-growe, 2021-06-01:
Ganindu,

This will not work with JetPack 4.4. 

As it turns out, for an unrelated reason, I uploaded branches named d3/dev/jp-4.5.1 to all of the submodules repositories. While D3 doesn't officially support JetPack 4.5.1 you can update if you do the following:

1. Update your l4t-submodules directory to d3/dev/jp-4.5.1 and then do a submoodule update for all child sumodules (` git submodule update --init --recursive` will do it)
2. Update to the tegra_defconfig for the kernel and configure the kernel to your liking
3. Rebuild

We're busy working on JetPack 4.5.1 support at D3 and this *should* bring you mostly up to date with that work. There are no promises that it will work though. 

Greg

ganindu7, 2021-06-01:
Hi Greg,

Thanks a lot for the update (Happy memorial day)! 

Here is an update:

###Jetpack 4.4:

[These instructions](https://github.com/D3Engineering/d3-jetson-bsp/tree/d3/4.0.0) do work in build > sync > reboot > login into os (where I am at)

### Jetpack 4.5.1

* I did the submodule folder switch and child submodule update as you mentioned. 
* I used the **`make linux-defconfig`** command (Is there a way to specify the tegra_deconfig in another way?) 

it built > sync > reboot > (after that was stuck on the D3 boot screen)

I tried several times and got the same result, But I have a suspicion that there might be hard coded 4.4 specific stuff in whatever generated the configure script?  

If It's too tricky to fix 4.5.1 let me know I'll make do with 4.4 and wait for 4.5.1 release. 

Cheers,
Ganindu.

D3-growe, 2021-06-01:
Ganindu,

In this case you'll need to debug deeper or wait for JP 4.5.1 support from D3. The defconfig pulls in D3's default configuration which hasn't been updated for JP 4.5.1 on GitHub. The basic technique is to pull in NVIDIA's tegra_defconfig and then modify it to add what you need.

Thanks,
Greg

ganindu7, 2021-06-01:
Hi Greg, 
Thanks I will do that. Just one last question. Once I've got this working (I think now it's working for 4.4) I want to see the camera video. 

Is there a script somewhere I need to run? host or target? (e.g. do I need to build  `d3-jetson-util` in the target and run select camera script?) Is there a documentation site for a D3 newbie to use the 16-camera board? I got [this camera ](https://www.d3engineering.com/product/designcore-d3rcm-ov10640-953-rugged-camera-module/) and  [board](https://www.d3engineering.com/product/designcore-nvidia-jetson-agx-xavier-interface-card/) 

Cheers,
Ganindu.

closed 2021-06-01 by ganindu7

D3-growe, 2021-06-01:
Ganindu,

In theory this is documented in the D3 "release notes" that serve almost as a manual. We recognize that the quality of that document is not really where we would like it to be.

The short version for what you need to do is:
1. Install the kernel and utils package using apt
2. Install the dtb for `d3-xavier-2xocto.dtb` using `d3-install-dtb`. The 2xocto name is something we used internally to refer to the 16x fpdlink interface board. We should have renamed it in the code to something less confusing but, alas, we did not.
3. Reboot
4. Run `d3-select-cameras-boot` to configure the ov10640s on the ports you want. This script edits /boot/extlinux/extlinux.conf to add a kernel parameter named `active_overlays`. You could also use a text editor to add the correct values yourself. Some people prefer that.
5. Reboot
6. Check that /dev/video0 (and other, possibly) exist
7. Run `d3-build-tegra-media-api`. Once built run `argus_camera`. This is an example program from NVIDIA and the source code is included in their package. Our script only builds and installs the code. It serves as a good example for how to program using the libargus API.

Regards,
Greg

ganindu7, 2021-06-01:
That step 1. for the debian version right? because I built the kernel `uname -a ` returns `Linux Jetson 4.9.140+ #1 SMP PREEMPT Tue Jun 1 13:36:32 BST 2021 aarch64 aarch64 aarch64 GNU/Linux`

`cat /proc/device-tree/nvidia,dtsfilename` returns `arch/arm64/boot/dts/../../../../../../hardware/d3/d3-xavier-2xocto.dts`

I'm not sure how to check if the utils were installed on the target (make sync)  too, but I can see that automake has generated the   Makefile with relevant fields.

D3-growe, 2021-06-01:
Building the kernel is fine and it looks like you've got the correct kernel and device tree.

The utils are not installed via sync. I recommend installing the .deb from D3's binary package (you can download it from GitHub).

You should be fine and not need to revert anything. The way you are doing this is how I work with the systems here. :)

ganindu7, 2021-06-01:
Thanks Greg!, that is fantastic to hear!!

ganindu7, 2021-06-02:
@D3-growe It seems like the .deb for the utils isn't on the public repo.

D3-growe, 2021-06-02:
It is tucked inside here: https://github.com/D3Engineering/d3-jetson-bsp/releases/download/D3%2FRELEASE%2F4.0.0/d3-jetson-bsp-4.0.0.zip

If you unpack the .zip you'll find a source archive and a binary archive. You'll find the utils .deb inside of the binaries archive.

ganindu7, 2021-06-03:
Thanks Greg! It's all working fine with Jetpack 4.4 :)    

I think for 4.5.1, the `d3_defconfig` parameters I have with the me needs changing to suit the update(I tried with the  latest `tegra_defconfig` but without the knowledge about the D3 setup comparing the two configs and do trial and error until everything works seemed a bit too involved for now)   as you explained earlier , I will wait till you sort the d3_config for 4.5.1 out. 

Please do give a shout when that's available. 

Thanks for the quick responses and all the encouraging words Greg!! 
Ganindu.