D3Engineering/d3-jetson-bsp #15: d3-install-dtb

qodlsrud20712, 2021-07-04:
what is Select a valid d3 bsp kernel release?

greg-rowe, 2021-07-06:
When you run `d3-install-dtb` you will first be presented with a menu of kernels that are installed on the filesystem. "Select a valid d3 bsp kernel release" means to choose, from that menu, a D3 kernel. Once that is chosen you will see a second menu with options for which .dtb file to install.

D3-growe, 2021-07-06:
When you run `d3-install-dtb` you will first be presented with a menu of kernels that are installed on the filesystem. "Select a valid d3 bsp kernel release" means to choose, from that menu, a D3 kernel. Once that is chosen you will see a second menu with options for which .dtb file to install.

closed 2021-07-20 by D3-growe