D3Engineering/d3-jetson-bsp #16: Camera images from D3 Hardware > libargus > GStreamer pipeline

ganindu7, 2021-08-15:
Hi, 

I'm trying to make a DL pipeline for an automotive product and it seems using  GStreamer is the most suitable. Now I'm pretty familiar with the argus capture control API, EGLStreams and nv_buffers. I think it makes sense to create a gstreamer srcbin with argus as the image provider (even potentially with feedback paths to control sensor/ISP parameters on the fly). I'm just wondering before committing if there are better options or there is any guidance available? for example I know `nvarguscamerasrc` source is public . but I'm not sure where the files are?.  My thoughts are based on the  [camera architecture][nv-cam-aarch] diagram published by nvidia.

Extra question: 
currently working with the D3 16 camera board and it's quite great! However sometimes it doesn't recognise the cameras at boot and sometimes I need to restart/move/wiggle things around (if it happens usually happens once a day) we don't mind if this is purely a dev board connection issue (which we can fix in a more production ready form factor I assume) ? 
Cheers,
Ganindu.

[nv-cam-aarch]: https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3231/Tegra%20Linux%20Driver%20Package%20Development%20Guide/images/image4_7.png

D3-growe, 2021-08-17:
Hi Ganindu,

I'll try to cover everything but you have a lot in here! If I miss anything please remind me and point it out!

* Yes, I'm D3-growe on the developer forums. D3 is many more people than just me, of course, but that is me.
* In many cases nvarguscamerasrc as the source in a GStreamer application is a very good choice. There are some features that you may desire but, the source is available so, depending on the licensing you may be able to adapt it. D3 has had a case or two where nvarguscamerasrc was missing a key feature so we developed our own. At that time the source for  nvarguscamerasrc was not available. If you have specific needs D3 may have already developed something that would suit you.
* The issue you are having with D3 ov10640 is a known issue. We know what is wrong -- it's the i2c address strapping pins on the image sensor inside of the camera module -- but we don't have a fix developed yet. We also know how to fix it but haven't yet. We will be having a new mainline BSP release and will certainly include a fix for that in the next release (D3 BSP 5.0.0)
* Please email me directly about your needs that nvarguscamerasrc might not meet (twiddling gains, etc). I will try to answer your questions via email. Our "d3 argus camera src" or DRACS, might do what you need.
* To open multiple cameras with nvarguscamerasrc use the sensor-id property. If you look at /etc/udev.d you'll find an example rules file that you can use to map video nodes (/dev/video$N) to physical ports. The sensor-id parameter will always increment from 0-N so, if your cameras are always connected in the same way then you will always have the same sensor-id

Again, please let me know if you have further questions and don't hesitate to email me if you'd like.

Thanks!
Greg

ganindu7, 2021-08-18:
Hi Greg, 
[context]
Thanks a lot for getting back to me. I've resolved almost all problems on my end except for the ISP tweaking on the fly with GStreamer, most of the starting pain is due to the fact that I'm quite new to perception (coming from a controls background) and having to learn the concepts by mostly reading source-code (because I couldn't find any other definitive guide of sorts), 

In the very beginning I started building a pipeline with VX  and realised it's not really useful because I couldn't get the camera working with a simple URI then,  As the D3 docs stated (once I got access) I started with the argus camera app (basically kind of reverse engineered it) Then moved onto EglStreams and unix sokets  with the aim for process separation in the pipe (including writing shaders haha) to build a third pipeline only to realise If I'm using shaders (compute shaders) I'll be missing out on a lot of developments over the years and basically will have to reinvent the wheel (and other bits too). 

Then I Came across gstreamer because most useful  DL pipelines use that, because I felt that the stock nvidiaGstreamersrc binary is broken and unsuitable, I did some digging, found the source did the modifications worked with the accelerated gstreamer development guide and right now I think I'm in a good place having having quite convinced that this is the right way to go ahead, thanks a lot for the reassurance because I pretty much feel I've tried all other paths :').

[about the camera issue ]
That is fine and we are happy to work with D3 to make production units upto the level where there won't be pain for the end users  and to us (having to provide support/replacements)

[back to `nvarguscamerasrc`] 
I will use the stock source for now and get the ground running (I'm basically learning gstreamer simultaneously too) because then I wouldn't have to  take your time doing the basic stuff. Once I get to the level I'd like to close loop control ISP/Camera stuff I will get back to you. 

[about camera picture]
This is something that I can't figure out how it happened. When I got the D3 kernel to work last time the image was like the picture below ![previous_image]  

Then one day I got an Ubuntu update (probably a kernel one I OK'd without checking) It did something which caused the entire gui to disappear but I could ssh into it and that was the only way and after I disabled the X server I could log in with headless mode. 

Then I re-flashed Jetson (I noticed the 4.4.1 in SDK manager and tried it, got the same result. then went back to 4.4) Immediately after reflash the stock argus camera app was crashing (I couldn't run `argus_camera`). However I had a binary from my reverse engineered arguscamera which worked. So I did some digging in the forums and found out that I had to add a source [patch][argus-patch] to make it work. (Interestingly I have no recollection of doing that the previous time) , However After that exercise I feel like the Image quality has deteriorated. both images below are after the update.

This image was taken in the evening after the update
![washed_out_image]

This was taken today 18 Aug around 10am.

![image-today]

I'm not exactly sure what's going on but I am quite confidant that it looks less vibrant under any lighting condition compared to before the update. (I did wipe the lens too haha)

tbh this will not affect my current work but I thought it's best if I let you know. 

Thanks again for getting back to me and it's a pleasure working with you!

Cheers,
Ganindu. 

[image-today]: https://user-images.githubusercontent.com/5096658/129869477-56cd6822-1b0b-4edd-9d64-a33ecbceff98.jpg

[washed_out_image]: https://user-images.githubusercontent.com/5096658/129860435-ce7e7a4e-511a-479d-859c-34a5f4877ae2.png

[previous_image]: https://user-images.githubusercontent.com/5096658/129866563-70e650c0-09fe-4f1d-bf16-e265f776e1d8.png

[argus-patch]: https://forums.developer.nvidia.com/t/l4t-r32-4-3-breaks-argus-camera-and-causes-segmentation-fault-sigsegv/141432/9

closed 2021-08-18 by ganindu7

D3-growe, 2021-08-19:
Ganindu,

You've had a very thorough journey in a very short time! Nicely done!

I have seen that 'unattended upgrades' can cause serious issues on Jetson. In more recent builds at D3 we've resorted to using apt pinning to prevent undesired upgrades from happening. Just as you saw the system could "upgrade" itself to a state that results in the entire GUI not working. If this were a 100% factory configured AGX Xavier I suspect that the upgrade would be fine. The problem, if I recall correctly, was an interface change and therefore mismatch between the D3 kernel and the GPU driver. I may have some of those details wrong.

As for the color issues that you are having please feel free to open a new bug report if you'd like. I suspect that the .isp file for your ov16040 cameras are not loading. First, look at /var/nvidia/nvcam/settings to ensure that the .isp files and a number of symlinks are present. If they are not present be sure to install d3-jetson-util package which contains the .isp overrides files. That package also configures the symlinks. You can also run nvargus-daemon in the foreground to examine which .isp overrides it loads. The dull output is strong evidence that the overrides are not being used.

Cheers,
Greg

ganindu7, 2021-08-20:
Hi Greg, 
That was indeed the ISP override file that was causing the flushed out picture. I installed it (d3_utils) from the debian file provided (not from this repo, it didn't build) 

One trap point (for anyone doing the same fix ) is that deb install resets the `/boot/extlinux/extlinux.conf` file which took me a while to figure out.

D3-growe, 2021-08-20:
Ah yes! I forgot about extlinux.conf being overwritten. We want to change our packaging to be a single .deb (just the kernel) and be more intelligent about handling extlinux.conf.

I'm glad to hear you are up and running properly. The ov10640 issue should be resolved in the next BSP release.

Good luck and don't hesitate to contact D3!

Thanks,
Greg