D3Engineering/d3-jetson-bsp #23: D3 Release 5.0.0 Not working with M.2 NvMe memory card attached to Xavier

ganindu7, 2022-06-20:
Hi, 
I'm using the 16 Camera board (FPD-LINK III) I can conform it is working with the new release but it is not booting with the M.2 Memory attached. Used to work fine with the 4.0.0 Release with JP 4.4.

Cheers,
Ganindu.

d3-cburrows, 2022-06-20:
Hello @ganindu7,

We have observed behavior similar to what you're describing in recent JetPack versions. The Xavier seems to try to boot to M.2 storage whenever it's connected, regardless of whether or not it is bootable. Take a look at the [NVIDIA docs](https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3271/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/flashing.html) to temporarily change the boot order.

If you're looking for a permanent solution, NVIDIA recommends [updating and re-flashing cbo.dtb](https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3271/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/getting_started.html). 

I hope this helps!

d3-cburrows, 2022-06-21:
Ganindu,

Interesting; thanks for bringing that up. We will have to investigate why installing our BSP would change the behavior. I can confirm that those documents still apply regardless of whether the BSP is installed.

ganindu7, 2022-06-21:
Hi Cody, 
Thanks a lot for getting back to me. Just a quick clarification. If I opted for the [[updating and re-flashing cbo.dtb](https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3271/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/getting_started.html) do I have to do something to retain the changes I already made with `sudo ./flash.sh -k kernel-dtb -d ./kernel/dtb/d3-xavier-16x-fpdlink.dtb jetson-xavier mmcblk0p1`. I'm asking this because I'm wondering if sudo ./flash.sh -k kernel-dtb -d ./bootloader/cbo.dtb might override the previous setting, is that fine or do I need to somehow combine the two strings?

d3-cburrows, 2022-06-21:
Hi Ganindu,

Good question. `sudo ./flash.sh -k kernel-dtb -d ./bootloader/cbo.dtb jetson-xavier mmcblk0p1` will overwrite the dtb for the linux kernel (in the `kernel-dtb` partition), and could render the Xavier unbootable. `cbo.dtb` should go somewhere else.

In addition, it might be a good idea to specify the `-r` option. Otherwise, `flash.sh` could shamelessly wipe everything to factory settings. I've been burned before!

You should write it to the `CPUBL-CFG` partition instead, like this:
`sudo ./flash.sh -r -k CPUBL-CFG -d ./bootloader/cbo.dtb jetson-xavier mmcblk0p1`

Cody

ganindu7, 2022-06-21:
Hi Cody, 
Strangely now everything works fine. 
what I did. 

Flashed the Jetson from SDK Manager to JP 4.6.1 (Note I tried my luck with with 4.6.2, didn't work )
Inside the bsp root dir
0. `make clean` (I also deleted the build dir just to be sure)
1.  `./bootstrap`
2. './configure --with-system-type=xavier --with-dtb='build/deploy/boot/d3-xavier-16x-fpdlink.dtb' --with-username=nvidia --with-target-host=Jetson-dev --with-l4t=/home/ganindu/nvidia/nvidia_sdk/JetPack_4.6.1_Linux_JETSON_AGX_XAVIER_TARGETS/Linux_for_Tegra
'
3. `make linux-defconfig`
4. make -j8
5. `./bin/prep-new-system-from-host` (I did the ssh-copy-id before as well, build mm-api for argus )
6.  `make-sync`
7. `make-reboot`
8. Voila!!!!!!!!!!!!!! It worked and I logged in
9.  edited `/boot/extlinux/extlinux.conf` to add camera info 
10. rebooted 
11. Tried argus_camera app 
12. All good!!
13. Then took the Jetson Xavier  apart and and installed the NvMe SSD, reconnected everything and started up 
14. EVERYTHING WORKS!!

But yesterday with the debian install I tried around 3 times with and without the NvMe SSD and It worked every time when it was not present! 

So I'm really sure what exactly caused it. 

Thanks a lot for the help!! Please let me know if you want anymore information to reproduce. 

P.S
 I tried  `sudo ./flash.sh -r -k CPUBL-CFG -d ./bootloader/cbo.dtb jetson-xavier mmcblk0p1` in one of the occasions today where I didn't get anything after the D3 screen on startup but nothing happened.  

Cheers,
Ganindu.

closed 2022-06-21 by ganindu7

FreddclrBM, 2022-06-28:
Hi,
Furthering this, I am also experiencing the same issue on Jetson AGX Xavier, 16 board camera controller while using M.2.
I haven't tried building from source also using the debian install.

Regards,
Freddie

d3-cburrows, 2022-06-28:
Thanks for letting us know, Freddie. I'll make sure this gets some attention before our next release!

Cody

Proinn, 2022-07-29:
Hello,

I am encountering the same issue when trying to install the BPS 5.0.0 D3 overlay on jetpack 4.6.1. I am also using a Jetson Xavier, 16 camera board and an SSD attatched to the Jetson.

Regards, 
Michiel