D3Engineering/d3-jetson-bsp #24: Camera picture issue (Intermittent)

ganindu7, 2022-06-24:
Hi, 
This It seems there is an intermittent issue where the picture come out with lots of lines. when I get this if I cycle through the camera modes the picture gets restored. 

[image]

[image]

However when it's good it's quite good.  

[image]

As a positive I haven't had a single crash with this release or a startup where cameras didn't work at all yet!! 

Cheers,
Ganindu.

ganindu7, 2022-07-05:
> Ganindu,
>
> Sounds like we're dealing with the same issue here. Unfortunately, all we
> know for sure is that covering the lens for a few seconds resolves the
> noise. This is on our list for the next release.
>
> One thing I just thought of: are the results the same if you disable ISP
> features like auto exposure, auto white balance, edge enhance, etc.? gst-inspect-1.0
> nvarguscamerasrc should tell you how to disable them.
>
> Cody

Hi Cody,
Thanks for the tip!
The easy fix is shuffle the camera mode, I will hopefully try the
suggestions and let you know.

We are currently doing some field testing so it might not be super quick!
Thanks again for getting back to us!

Cheers,
Ganindu

d3-cburrows, 2022-10-20:
Hi Ganindu,

We finally had some time to dive into this, and determined that we're running a clock in the image sensor outside its published operating range. That would explain the intermittent nature of the artifacts. Unfortunately it's not a straightforward fix so we've reached out to OVT for assistance. At the moment, the best known workaround is to switch modes as you mentioned.

Cody

ganindu7, 2022-10-20:
Thanks Cody! Thanks a lot for remembering!!

I thought it might be a PCB noise coupling issue or something like that
causing sync issues.

When not controlled automatically manually covering and opening the imager
helps too as you let us know.

Cheers,
Ganindu.