D3Engineering/d3-jetson-bsp #25: AGX Orin support

ganindu7, 2022-08-11:
Hi I'm using the D3 16 camera board,  
Can I use this BSP(5.0.0/with jetpack 4.6.1) with the new Jetson Orin Developer kit? 

Cheers, 
Ganindu.

ganindu7, 2022-08-11:
Just noticed that I can only install JP 5.0.0 (Without deepstream) / 5.0.1 DP (with deepstream) so I guess my chances are quite slim :'(

d3-cburrows, 2022-08-11:
Hi Ganindu,

Thanks for your question. Unfortunately no, this release can't be used on the Orin devkit. On the hardware side, the 16-channel camera board connector is not tall enough to physically clear the taller USB ports the Orin devkit has. You may not be able to fully seat the connector or utilize all of the camera ports. We are working on new software and hardware to support Orin that will hopefully be available soon.

Thanks,
Cody

closed 2022-09-28 by d3-cburrows