D3Engineering/d3-jetson-bsp #27: ln: **failed to create hard link '/boot/active-dtb' => './d3-xavier-6x-serdes.dtb': Operation not permitted**

vivekJB1, 2022-09-26:
Hi @D3-growe , @D3-jwatts , @d3-sdunnington ,

I am trying to flash D3 Jetson SerDes Sensor Interface Card with  D3 Jetson BSP v5.0 on a target device (Jetson Xavier AGX).

Configuration:
1> Target Device : Jetson Xavier AGX
2> D3 Hardware to flash: D3 Jetson SerDes Sensor Interface Card
3> D3 BSP: D3 Jetson BSP v5.0 
4> Hardware Setup: D3 Jetson SerDes Sensor Interface Card is mounted on Jetson Xavier AGX and power is ON.

Procedure Implemented:
Step 1: sudo rm -f /etc/dpkg/dpkg.cfg.d/excludes

Step 2:  sudo apt update

Step 3: sudo apt install \
./linux-image-4.9.253+5.0.0-d3+_1_arm64.deb \
./d3-jetson-util_5.0.0-1_arm64.deb

At the end of step 3, following error observed: "d3-install-dtb] failed to link dtb! (file ./d3-xavier-6x-serdes.dtb)"

Step 4: uname -a
Linux jayben-desktop 4.9.253-tegra #1 SMP PREEMPT Sun Apr 17 02:37:44 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux

Question to D3: Is the BSP installed successfully? As mentioned above, uname -a  shows "4.9.253-tegra #1 SMP PREEMPT Sun Apr 17 02:37:44 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux".

How can I verify that BSP is successfully installed? 

-------- Below is my terminal log for your analysis--------

jayben@jayben-desktop:~$ 
jayben@jayben-desktop:~$ uname -a
Linux jayben-desktop 4.9.253-tegra #1 SMP PREEMPT Sun Apr 17 02:37:44 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux
jayben@jayben-desktop:~$ d3-install-dtb
[d3-install-dtb] Choose an installed D3 BSP release
1) ./linux-image-4.9.253+5.0.0-d3+
#? 1
[d3-install-dtb] Choose a dtb to install
1) NVIDIA Jetson NX on DesignCore NVIDIA Jetson Xavier NX 12-Camera Carrier Board (d3-nxc-fpdlink.dtb)
2) NVIDIA Jetson TX2 with DesignCore NVIDIA Jetson SerDes Sensor Interface Card (d3-tx2-6x-serdes.dtb)
3) DesignCore NVIDIA Jetson RSP-TX2 FPD-Link (d3-tx2-rsp-fpdlink.dtb)
4) NVIDIA Jetson AGX Xavier with DesignCore NVIDIA Jetson AGX Xavier FPD-Link III Interface Card (d3-xavier-16x-fpdlink.dtb)
5) NVIDIA Jetson AGX Xavier with DesignCore NVIDIA Jetson SerDes Sensor Interface Card (d3-xavier-6x-serdes.dtb)
#? 5
ln: **failed to create hard link '/boot/active-dtb' => './d3-xavier-6x-serdes.dtb': Operation not permitted**
[d3-install-dtb] failed to link dtb! (file ./d3-xavier-6x-serdes.dtb)
jayben@jayben-desktop:~$ uname -a
Linux jayben-desktop 4.9.253-tegra #1 SMP PREEMPT Sun Apr 17 02:37:44 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux
jayben@jayben-desktop:~$

vivekJB1, 2022-09-26:
jayben@jayben-desktop:~$ cat /proc/device-tree/nvidia,dtsfilename
/dvs/git/dirty/git-master_linux/kernel/kernel-4.9/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t19x/galen/kernel-dts/common/tegra194-p2888-0001-p2822-0000-common.dtsijayben@jayben-desktop:

d3-cburrows, 2022-09-26:
To figure out whether or not the BSP was successfully installed, you will have to reboot then run `uname -a`, like you did. `4.9.253-tegra` indicates that you're still running the NVIDIA kernel. The D3 kernel will report `4.9.253+5.0.0-d3+` in its place.

The DTB step is simply creating a hard link to the DTB file called `/boot/active-dtb`. The `/boot` directory is owned by root, so you most likely don't have permission to write there as a standard user. Try again with super user privileges by running `sudo d3-install-dtb`. If it succeeds, you will need to reboot to use the new kernel and DTB.

Other than that, it looks like you're following the correct procedure. Let me know how it goes.

Thanks,
Cody

vivekJB1, 2022-09-27:
Hi @d3-cburrows,
Thanks for validating and verifying my executions.... I finally manage to install D3 BSP v5.0.

But, once installed, I tried to run nvgstcapture "as following:
nvgstcapture --sensor-id $0
But no camera got activated. I even tried changing ports from 0 to 5 and everytime it throws an error message as shown below:
------- Terminal log---------------------------
jayben@ubuntu:/home$ nvgstcapture --sensor-id $0
Encoder null, cannot set bitrate!
Encoder Profile = High

** (nvgstcapture:16197): ERROR **: 15:40:21.844: <main:4541> option parsing failed: Cannot parse integer value ?bash? for --sensor-id
Trace/breakpoint trap (core dumped)
-----------------------------------
Can you suggest what can I do to activate my camera?
I even performed "Camera Selection Utility" by executing d3-select-camera-boot and set all ports 
[image]

Can you suggest what can I do to activate my camera?

d3-cburrows, 2022-09-28:
It looks like we have two issues here. I want to try to address them separately. The first is to figure out whether or not your cameras are correctly detected and probe. The second is to figure out why `nvgstcapture` is failing.

To address the question of whether or not your cameras are present is to list video nodes in `/dev/`. For instance, If you have 5 cameras plugged in and enabled (it looks like you do, according to your screenshot) you should see 5 video nodes when you run `ls /dev/vid*`: `/dev/video0` through `/dev/video4`. Make sure to double check that the type of cameras you have connected matches your configuration in `d3-select-cameras-boot`. Once you have video nodes, a simple way to test streaming from them is to use `v4l2-ctl --stream-mmap` (from the `v4l-utils` package). You should see a bunch of `<` characters printed on the terminal if they're working. I'd recommend making sure that you can stream with `v4l2-ctl` before getting `nvgstcapture` working.

It looks like you're passing `$0` as the argument to `nvgstcapture`. Bash treats that as a special variable, not the number 0. To see what I mean, see what `echo $0` says. Double-check the argument you mean to pass in and see if that helps.

Thanks,
Cody