D3Engineering/d3-jetson-bsp #28: Help required to install BSP in my Jetson Xavier AGX  for SerDes sensor interface card

vivekJB1, 2022-09-26:
Hi @D3-growe  @D3-jwatts  @d3-sdunnington ,

**Problem Statement:** How to flash SerDes Sensor Interface Card  with D3 BSP v5.0 on  Jetson Xavier AGX?

**Configuration:**
1> Target Device : Jetson Xavier AGX (Jetpack 4.6.1)
2> D3 Hardware to flash: D3 Jetson SerDes Sensor Interface Card
3> D3 BSP: D3 Jetson BSP v5.0
4> Hardware Setup: D3 Jetson SerDes Sensor Interface Card is mounted on Jetson Xavier AGX and power is ON.

**Pre-requisites:** 
1> Target Device Jetson Xavier AGX is flashed with Jetpack 4.6.1.
2> SerDes card is mounted on Jetson Xavier AGX.
3> SerDes interface Card jumpers J13 and J14 are connected to 12V.
4> Power supply is ON for Jetson Xavier AGX and SerDes Card.  SerDes LED D26, D27, D28, D29 are  solid ON (green).
5> Host PC is not connected. I am trying to flash SerDes via terminal on Target Device (Jetson Xavier AGX ).

**Description:**
Document (D3 Jetson BSP v5.0.0 Release Note) Provided is confusing and I am really struggling hard to flash  SerDes interface card with D3 BSP 5.0.

Requesting if you can help on how to flash it with correct procedure....

d3-cburrows, 2022-09-26:
I'm sorry to hear that you've found the Release notes to be confusing. We are always looking for feedback on how we can improve our documentation, so which steps in particular are the most confusing? I'd be happy to explain them. The installation instructions start in section 4.

vivekJB1, 2022-09-27:
Hi @d3-cburrows , After going through the doc couple of times I successfully installed D3 BSP v5.0.
Found small error in "D3-Jetson-BSP-Release-Notes.pdf", as shown below:

[image]

d3-cburrows, 2022-09-28:
Glad to hear it! Thanks for sharing that. We'll patch that up for the next release.

closed 2022-09-28 by d3-cburrows