D3Engineering/d3-jetson-bsp #29: Argus Installation on Jetson Xavier AGX

vivekJB1, 2022-09-27:
Hi @D3-growe @D3-jwatts  @D3-jwatts ,

This is regarding installation of Argus on Jetson Xavier AGX.
Can you confirm the below steps are correct to install Argus:

Step 1> Go to to Terminal.
S2> sudo apt update  # this will update the repositories.
S3> sudo apt install argus-server  # this will install the argus server on agx.
S4> d3-build-tegra-media-api
Note: Step 4 instruction will install Argus and message will appear “Terga Multimedia API Installed”.

S5> argus_camera&   #This step will execute/start Argus Camera Use.
        
But the argus_camera application did not started. Can you help how can I activate camera streaming using argus? Do I need to perform some extra setting for it?

d3-cburrows, 2022-09-28:
I think the `argus-server` package in step 3 is not what you're looking for. See this Ubuntu [package details](https://packages.ubuntu.com/bionic/net/argus-server) page.

Other than that, your steps look correct. Assuming that you've installed the SDK components from NVIDIA, and the `d3-build-tegra-media-api` script succeeded, you should have `argus_camera` somewhere in your path. Run `which argus_camera` to see for sure. You shouldn't have to do anything other than that.
[image]

Could you provide some more detail about what happens when you run `argus_camera`? Does the command exit with no output? Does it cause a segmentation fault? Are there any logs you could provide to help us narrow it down?

Thanks,
Cody

vivekJB1, 2022-10-18:
@d3-cburrows 
Hi Cody,
After installing d3 BSP 5.0 and argus, I rebooted system and camera started streaming well. 

But due to other circumstances, I re-installed jetpack 4.6.1 and re-installed d3 BSP 5.0 and argus. But this time cameras are not streaming and throwing error. Can you help in this regards?

Below are the details:

**---------------------------------------Terminal -----------------------------**
jayben@jayben-desktop:~$ uname -a
Linux jayben-desktop 4.9.253+5.0.0-d3+ #2 SMP PREEMPT Fri Jun 10 11:20:17 EDT 2022 aarch64 aarch64 aarch64 GNU/Linux
jayben@jayben-desktop:~$ cat /sys/module/build_id/version
5.0.0
jayben@jayben-desktop:~$ cat /proc/device-tree/d3,jetson-bsp-build-id
66f41109-cf31-43ec-b942-01e9ee3e0af6

jayben@jayben-desktop:~$ cat /proc/device-tree/nvidia,dtsfilename
arch/arm64/boot/dts/../../../../../../hardware/d3/d3-xavier-6x-serdes.dtsjayben@jayben-desktop:~$ cat /proc/device-tree/nvidia,dtbbuildtime
Jun 10 202211:19:39

----------------------------------Error Details---------------------------------------------------
jayben@jayben-desktop:~$ **argus_camera**

Executing Argus Sample Application (argus_camera)
Argus Version: 0.98.3 (multi-process)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:957 No cameras available
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, Dispatcher:910 (propagating)
(NvCameraUtils) Error InvalidState: Mutex already initialized (in Mutex.cpp, function initialize(), line 41)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/ClientSocketManager.cpp, function open(), line 54)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 262)
(Argus) Error InvalidState: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 106)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:949 Failed to create CameraProvider
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, getInstance:933 Initalization failed
Argus Version: 0.98.3 (multi-process)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:957 No cameras available
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, getInstance:933 Initalization failed
(NvCameraUtils) Error InvalidState: Mutex already initialized (in Mutex.cpp, function initialize(), line 41)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/ClientSocketManager.cpp, function open(), line 54)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 262)
(Argus) Error InvalidState: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 106)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:949 Failed to create CameraProvider
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, getInstance:933 Initalization failed
Argus Version: 0.98.3 (multi-process)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:957 No cameras available
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, getInstance:933 Initalization failed
(NvCameraUtils) Error InvalidState: Mutex already initialized (in Mutex.cpp, function initialize(), line 41)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/ClientSocketManager.cpp, function open(), line 54)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 262)
(Argus) Error InvalidState: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 106)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:949 Failed to create CameraProvider
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, getInstance:933 Initalization failed
Argus Version: 0.98.3 (multi-process)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:957 No cameras available
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, getInstance:933 Initalization failed
(NvCameraUtils) Error InvalidState: Mutex already initialized (in Mutex.cpp, function initialize(), line 41)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/ClientSocketManager.cpp, function open(), line 54)
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 262)
(Argus) Error InvalidState: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 106)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:949 Failed to create CameraProvider
[... cut ...]
(Argus) Error InvalidState:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 262)
(Argus) Error InvalidState: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 106)
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, initialize:949 Failed to create CameraProvider
Error generated. /home/jayben/jetson_multimedia_api/argus/apps/camera/modules/Dispatcher.cpp, getInstance:933 Initalization failed
Segmentation fault (core dumped)