D3Engineering/d3-jetson-bsp #30: Distortion coeficient for IMX390 (1000843-R)

shams265, 2023-01-13:
Hi,
I am using your IMX390RCM sunex DSL239  192 HFOV (1000843-R) with DesignCore® NVIDIA® Jetson AGX Xavier FPD-Link™ III Interface Card. 
BSP using 5.0 version on Jetpack 4.6.1.
Hardware Jetson AGX Xavier Development Kit.

Problem Statement:
I am using Deepstream to live stream 7 cameras and successfully received them on the receiving end. After receiving i want to de-warp the received stream in order to get a flat 2D image. 
In order to correct the distortion and correctly configure nvdewarper element of deepstream, i need to know the exact distortion co-efficient s and camera matrix values as shown in picture below.
[image]
The distortion co-efficient s will be used for correction the nvdewarper configuration file, who has a template like below

[property]
#dewarp-dump-frames=10
output-width=3680
output-height=2428
num-batch-buffers=1

#########################################
# Note - Max 4 surfaces are supported
#########################################

[surface0]
# 1=PushBroom, 2=VertRadCyl, 3=PerspectivePerspective
projection-type=3
width=3680
height=2428
focal-length=1721.592
#src-fov=180
#top-angle=30
#bottom-angle=-30
**### distortion=-0.09787;0.00574;-0.00013;0;0**      (The distortion co-coefficient's values will go in here)
src-x0=1861.88
src-y0=1216.04
# 0=cudaAddressModeClamp, 1=cudaAddressModeBorder
#cuda-address-mode=0

------------------------------

Expected Support:
1> Please share these distortion coefficients to get flat 2D image?
2> Can we get any support on how to obtain flat 2D images out of IMX390 camera modules?

Any help will be highly  appreciated.
Regards

vivekJB1, 2023-01-13:
Following.....

D3-aavery, 2023-01-13:
Hi @shams265, please try the following camera intrinsics:
[IMX390_192HFOV.zip](https://github.com/D3Engineering/d3-jetson-bsp/files/10412544/IMX390_192HFOV.zip)
Note the change in resolution to what you show above. It also looks like nvdewarper expects the plumb_bob distortion model, however, equidistant provides better results on this ultra-wide field of view lens.

shams265, 2023-01-16:
Hi,
@D3-aavery i have consulted with one Nvidia's support team member and he has provided me the information stated below:
--------------------------------------------------------------------------
To get the correct distortion coefficients you need to use the following calibration method of opencv :
[https://docs.opencv.org/2.4.13.7/doc/tutorials/calib3d/camera_calibration/camera_calibration.html](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.opencv.org%2F2.4.13.7%2Fdoc%2Ftutorials%2Fcalib3d%2Fcamera_calibration%2Fcamera_calibration.html&data=05%7C01%7Csrathi%40nvidia.com%7Ceb422ffb6d0646de029508da85c47908%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C637969376148452994%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=VVq%2BLw%2B6UgSSGtrrBd4fHvsbw%2BrXB%2BcPYXsIfjMClrA%3D&reserved=0)
This will give you focal length(fx, fy), principal point(cx, cy) and 5 distortion coefficients : (k1, k2, p1, p2, k3)
You can set them in the dewarper plugin config file as follows :

focal-length=fx/fy/avg(fx,fy)
src-x0=cx
src-y0=cy
distortion=k1;k2;k3;p1;p2 (please note the reordering here)
------------------------------------------------------------------------------------------------------------
Since I am pretty new to this stuff, so can you please fill up/provide values for the entries mentioned in config file above.
Thanks and best regards

aja9675, 2023-01-16:
@shams265, for the data that I provided, here's the breakdown of components:
image_height = 1096
image_width = 1936
fx = 652.5771400378992
fy = 651.6950976471154
focal-length = avg(fx,fy) = 652.136118843
cx = 966.5794318241657
cy = 558.2694892223593
k1 = -0.3002562235144459
k2 = 0.0783848772023062
k3 = -0.008314964125280827
p1 = -0.0003913208983429832
p2 = 5.759098689473573e-05

jayben-2022, 2023-01-17:
Hi,
@D3-aavery, thank you so much for your timely and conclusive support. We have achieved the desired results using the values you have provided.
Once again thank you so much for your support.
Best Regards,
-Vivek and Shams

aja9675, 2023-01-17:
Excellent, I'm glad that it worked out!

D3-aavery, 2023-01-17:
Excellent, I'm glad that it worked out!

closed 2023-01-17 by D3-aavery