D3Engineering/d3-jetson-bsp #32: Connection with a self-designed Radar board

debrekXuHan, 2023-02-22:
Dear D3 Team, 

We have a self-designed Radar board which could output data by MIPI-CSI.
Is it possible that we connect this Radar board with NVIDIA Jetson with the interface card product.
We have purchased the interface card and where should we start?

d3-cburrows, 2023-02-22:
Hi @debrekXuHan,

Thanks for sharing about your radar board! Our interface cards are designed to be used with FPDLink or GMSL serializers, which take MIPI CSI input. Theoretically you could connect your sensor to a UB953, and create a Tegra camera platform driver for the sensor to stream data on the Jetson.

Unfortunately I won't be able to go into too many details about the implementation here. For more in-depth support, please contact sales@d3engineering.com.

Thanks,
Cody

debrekXuHan, 2023-03-06:
> Hi @debrekXuHan,
> 
> Thanks for sharing about your radar board! Our interface cards are designed to be used with FPDLink or GMSL serializers, which take MIPI CSI input. Theoretically you could connect your sensor to a UB953, and create a Tegra camera platform driver for the sensor to stream data on the Jetson.
> 
> Unfortunately I won't be able to go into too many details about the implementation here. For more in-depth support, please contact [sales@d3engineering.com](mailto:sales@d3engineering.com).
> 
> Thanks, Cody

@d3-cburrows 
Thanks a lot for your kindly reply.
I have one more quick question here. What if we have already connected the Radar sensor with UB953 serializer, do we still need to implement an extra platform driver? Cuz I noticed that there is the driver for UB953 in the BSP codes.

d3-cburrows, 2023-03-06:
@debrekXuHan,

Good question! You can make use of our existing UB953/UB960 drivers, though they may require modification to work in your particular case. You would still need to implement a Tegra camera platform driver to enable the Jetson to ingest the CSI data from your sensor.

Here is NVIDIA's documentation about writing that kind of driver: https://docs.nvidia.com/jetson/archives/r35.2.1/DeveloperGuide/text/SD/CameraDevelopment/SensorSoftwareDriverProgramming.html

Our camera drivers all adhere to that framework, so they could be used as an example or a starting point for you. 

I hope that clears it up!

Cody