D3Engineering/d3-jetson-bsp #34: DS90UB960 output abnormal voltage during CSI2 HS transmission

DesWudio, 2023-05-25:
Dear team,

We are using the [NVIDIA Jetson SerDes Sensor Interface card](https://www.d3engineering.com/product/designcore-nvidia-jetson-serdes-card/) and it was connected to an Jetson AGX Orin. A UB953 serialiser is used to transfer the CSI2 signal (4096*512, 10FPS, RAW8, 400Mbps, 4lane+1CLK) from the sensor to the FPD3. 

We connect only one sensor and the UB960 outputs the stream on port0. The correct frame size was obtained from the UB960 registers on the interface card. No registers that indicate errors had been set.

However, we got "LP Sequence Error" ("status:0x00000088" and "status:0x00000080") when trying to use V4L2 to read the CSI2 input stream on Orin.

We had checked 2 things by using an oscilloscope:
- We checked the signal global time parameters from the wavefrom of the **UB960 CSI output pins**, they are all within the required range as defined in the MIPI Alliance Specification for D-PHY. 
- We have calculated the time it takes to transmit our 4096 byte frame line using single long packets per lane, the actual time taken in the waveform for long packet transfers was the same. 

But one strange thing we found from the waveform was the voltage. 

The signal voltage level on the data lanes during high-speed transmission ranges from 0-500mV, but when it has switched to low-power mode and the voltage is in the normal range (1.2V).

We wonder if this voltage problem is due to the fact that we didn't choose the right place (960 CSI2 out pins directly) to test the signals?

And could you please give us some ideas on how to overcome the errors that we have encountered and how to read the stream correctly at the end of Orin?

Best,
Lee

d3-cburrows, 2023-05-25:
Hello Lee,

Please send this inquiry to support@d3engineering.com so it can be routed to someone who could answer your hardware questions better than me.

The CSI error messages that the Orin SoC generates can be frustrating and difficult to understand. Unfortunately I don't know what LP Sequence Error means. NVIDIA would be the best resource to determine what the best course of action to correct that error is. I apologize that I could not be of more assistance.

Thanks,
Cody

closed 2023-05-25 by d3-cburrows

DesWudio, 2023-05-26:
Cody, thank you for your information.

DesWudio, 2023-06-07:
> Hello Lee,
> 
> Please send this inquiry to [support@d3engineering.com](mailto:support@d3engineering.com) so it can be routed to someone who could answer your hardware questions better than me.
> 
> The CSI error messages that the Orin SoC generates can be frustrating and difficult to understand. Unfortunately I don't know what LP Sequence Error means. NVIDIA would be the best resource to determine what the best course of action to correct that error is. I apologize that I could not be of more assistance.
> 
> Thanks, Cody

@d3-cburrows  

Dear Cody,

I have sent a e-mail to [support@d3engineering.com](mailto:support@d3engineering.com) several days ago, however, I still failed to receive any response. The title of the mail is "DS90UB960 output abnormal voltage during CSI2 HS transmission".

Would you please help us to push it?

Thanks in advance.