D3Engineering/d3-jetson-bsp #37: 6.1.0

GrigoriyPeters, 2023-09-12:
Hello,
I tried version 6.1.0  and encountered not working tegra-xusb on Jetson Xavier NX dev-kit compatible board:

```
[    3.524474] tegra-xusb 3610000.xhci: Direct firmware load for nvidia/tegra194/xusb.bin failed with error -2
[    3.524496] tegra-xusb 3610000.xhci: Falling back to sysfs fallback for: nvidia/tegra194/xusb.bin
```
And it was fixed when I added initrd line in extlinux.conf file from original Jetpack 5.0.2 extlinux entry.
But I couldn't find a correct way to build initrd with your bsp.
Or maybe I do something wrong.

What I did:
1. Install default Jetpack 5.0.2 via Nvidia SDKManager - everything works fine
2. Download d3/6.1.0  and configure it with line:
```
./configure --with-system-type=nx --with-dtb="$(pwd)/build/deploy/boot/d3-nxc-emmc.dtb" --with-target-host="jetson-usb"
```
3. Build it with default kernel config:
```
make linux-defconfig && make -j8
```
4. Download it:
```
make sync
```
5. Put an extlinux entry for this:
```
LABEL d3
    MENU LABEL D3
    LINUX /boot/d3-image
    FDT /boot/d3-nxc-emmc.dtb
    APPEND ${cbootargs} root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyTCU0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0 nv-auto-config
```
6. Reboot it

P.S.: Everything was fine on version 4.0.0 and dev/jp4.6

Best regards,
Grigoriy

d3-cburrows, 2023-09-13:
Hi @GrigoriyPeters, do you have `/boot/initrd` on your system? If so, you should be able to add a line to your `extlinux.conf` to load it as you mentioned. For example:

```
LABEL d3
    MENU LABEL D3
    LINUX /boot/d3-image
    INITRD /boot/initrd
    FDT /boot/d3-nxc-emmc.dtb
    APPEND ${cbootargs} root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyTCU0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0 nv-auto-config
```

We don't modify the NVIDIA provided ramdisk, so it should work fine for you. It's in `$L4T/bootloader/l4t_initrd.img`. You'll see we reference it in `Makefile.am` and `bin/builddeb-hook` during builds. The reason we don't rebuild it is because it contains firmware we don't have access to. `tegra-xusb` is an example.

I hope that helps!

GrigoriyPeters, 2023-09-14:
Hello @d3-cburrows 
Thanks for the reply! As I mentioned before, I fixed it the same way, but I was thinking of it as a hack, not a solution.
But if the hack is good enough, then it's the solution.

closed 2023-09-14 by GrigoriyPeters